gazoodle/CenturyVSPump #1: Assistance with Jandy VSPlusHP Pump

dannytrigo, 2023-01-25:
Hi,

I'm trying to use your module to control a Jandy pump.  I'm using a NodeMCU (ESP8266) and was trying to use its UART tx/rx pins, connected to my pump.  However, I now think I need an RS485 interface module between the ESP8266 and the pump, converting from serial tx/rx to RS485 A/B pins - can you confirm?

Sorry for opening an issue for this as its not an issue with your module.

gazoodle, 2023-01-26:
Yes, you'll need an RS485 TTL converter. On your MCU, the RX/TX pair are a full-duplex communication system, but RS485 is half-duplex over two wires. It uses a crossing system where both lines are needed for transmission from either the controller, or the device and so you need some trivial logic to make this transition. You can build it yourself, but it's so cheap to get. There are literally dozens of them available from almost all electronic supply shops or even Amazon. I personally love the ones from M5Stack because most of them come with a DC-DC buck converter which means you can power your ESP from the RS485 port on the pump.

closed 2023-01-26 by gazoodle

dannytrigo, 2023-01-27:
Thanks so much! Ordered