gazoodle/CenturyVSPump #2: Code stopped working after ESPHome update

qwest07, 2023-04-02:
Hello,

Code seems to stop working after ESPHome update.
Looks like ESPHome did incompatible update, modbus_controller was introduced.

Any chance you can reveiw changes?

Thank you!

gazoodle, 2023-04-03:
Not near my PC until later this week, I’ll take a look then.

gazoodle, 2023-04-06:
I've updated to the latest ESPHome 2023.3.2, recompiled from clean with both external components:

```python
external_components:
  # Needed to implement latest modbus user-defined function handling
  - source:
      type: git
      url: https://github.com/gazoodle/esphome
      ref: bug-fix-modbus-user-defined-function-handling
    components: [modbus]
  # Location of CenturyVSPump component implementation
  - source:
      type: git
      url: https://github.com/gazoodle/CenturyVSPump
      ref: main
```
and just the CenturyVSPump code:
```python
external_components:
  # Location of CenturyVSPump component implementation
  - source:
      type: git
      url: https://github.com/gazoodle/CenturyVSPump
      ref: main
```
Both times the code functions as expected, there are no compile faults and the code talks to the pump and I'm able to set the pump speed correctly. Not sure what is happening for you? Perhaps a clean build needed if there are some old files hanging around. Maybe add some logs to this issue while you're trying to control the pump?

qwest07, 2023-04-07:
Thank you for quick reply.

I did clean build - still having same issues - basically no communication with pump.

Here is part of the log:

18:56:40	[D]	[uart_debug:158]	
>>> "\x15E \x00\x03}\x04"
18:56:40	[D]	[uart_debug:158]	
>>> "\x15E \x00\x03}\x04"
18:56:41	[D]	[uart_debug:158]	
>>> "\x15E \x00\x03}\x04"
18:56:41	[D]	[century_vs_pump:140]	
Pump command 45 no response received - removed from send queue
18:56:41	[D]	[uart_debug:158]	
>>> "\x15E \x00\x00=\x05"
18:56:41	[D]	[uart_debug:158]	
>>> "\x15E \x00\x00=\x05"
18:56:41	[D]	[uart_debug:158]	
>>> "\x15E \x00\x00=\x05"
18:56:41	[D]	[uart_debug:158]	
>>> "\x15E \x00\x00=\x05"
18:56:42	[D]	[uart_debug:158]	
>>> "\x15E \x00\x00=\x05"
18:56:42	[D]	[century_vs_pump:140]	

Question:
- Do we still need to include  bug-fix-modbus-user-defined-function-handling? I think  it is part of ESPHome already
- Do we need to configure also modbus_controller?  According this: https://esphome.io/components/modbus_controller.html
-

qwest07, 2023-04-21:
Example file example_century_vs_pump.yaml is not building anymore after ESPHome updated to  2023.4.0

Here is build log:

In file included from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp:1:
src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h: In constructor 'esphome::century_vs_pump::CenturyVSPumpSensor::CenturyVSPumpSensor(esphome::century_vs_pump::CenturyVSPump*, uint8_t, uint8_t, uint16_t, const string&, const string&)':
src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h:22:30: error: no matching function for call to 'esphome::century_vs_pump::CenturyVSPumpSensor::set_name(const string&)'
                 set_name(name);
                              ^
In file included from src/esphome/components/sensor/sensor.h:5,
                 from src/esphome/components/centuryvspump/CenturyVSPump.h:7,
                 from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h:3,
                 from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp:1:
src/esphome/core/entity_base.h:20:8: note: candidate: 'void esphome::EntityBase::set_name(const char*)'
   void set_name(const char *name);
        ^~~~~~~~
src/esphome/core/entity_base.h:20:8: note:   no known conversion for argument 1 from 'const string' {aka 'const std::__cxx11::basic_string<char>'} to 'const char*'
*** [/data/century-vs-pump-controller/.pioenvs/century-vs-pump-controller/src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp.o] Error 1

Can you verify?

Thank you!

rx78nt1alex, 2023-04-25:
Can confirm that I receive the same error. Logs below

/Compiling /data/pool-pump/.pioenvs/pool-pump/src/esphome/components/network/util.cpp.o
In file included from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp:1:
src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h: In constructor 'esphome::century_vs_pump::CenturyVSPumpSensor::CenturyVSPumpSensor(esphome::century_vs_pump::CenturyVSPump*, uint8_t, uint8_t, uint16_t, const string&, const string&)':
src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h:22:30: error: no matching function for call to 'esphome::century_vs_pump::CenturyVSPumpSensor::set_name(const string&)'
                 set_name(name);
                              ^
In file included from src/esphome/components/sensor/sensor.h:5,
                 from src/esphome/components/centuryvspump/CenturyVSPump.h:7,
                 from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h:3,
                 from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp:1:
src/esphome/core/entity_base.h:20:8: note: candidate: 'void esphome::EntityBase::set_name(const char*)'
   void set_name(const char *name);
        ^~~~~~~~
src/esphome/core/entity_base.h:20:8: note:   no known conversion for argument 1 from 'const string' {aka 'const std::__cxx11::basic_string<char>'} to 'const char*'
*** [/data/pool-pump/.pioenvs/pool-pump/src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp.o] Error 1
========================= [FAILED] Took 17.90 seconds =========================/

gazoodle, 2023-04-25:
Ok, latest code fixes the set_name problem. I didn't check why this happened, but I'm guessing that the ESPHome base code deprecated the overload that used to take a `std:string`. Using the `.c_str()` method will defer to the `const char *` overload instead.

If you still get the problem within 24 hours of this comment, then you might need to temporarily changed the refresh interval like this ...

```yaml
external_components: 
  - source: 
      ref: main
      type: git
      url: https://github.com/gazoodle/CenturyVSPump
    refresh: 0s

```

qwest07, 2023-04-25:
Code is building now - thank you!

qwest07, 2023-04-26:
Closing since issue solved

closed 2023-04-26 by qwest07