gazoodle/CenturyVSPump #4: Cannot see the rpm value

fgamache1, 2023-05-08:
Hi. 
Thank a lot for the development of that. 

I have a problem.
I cannot see the rpm value. The button and the number selector work. I can set my rpm by pressing the button, or sliding with the slider. But the slider not showing the right number in the value, and I have a Unknow value to rpm sensor.

My pump is a Pentair VS Superflow. I try with the dipswitch 1 or 5 at ON. And both have the same result.

Can you help me ?

Thanks

[image]

[image]

gazoodle, 2023-05-09:
Can you post some logs from the device after startup? You should be able to get them from the ESPHome dashboard, use View Logs and then restart the device using the Restart button in HA. Then download them and add them to this issue.

fgamache1, 2023-05-09:
I have some other sensor, like a flow meter and a pressure meter too.

'
INFO Reading configuration /config/esphome/esphome-web-5c8418.yaml...
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-5c8418 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/esphome-web-5c8418/.pioenvs/esphome-web-5c8418/src/main.cpp.o
Linking /data/esphome-web-5c8418/.pioenvs/esphome-web-5c8418/firmware.elf
RAM:   [=         ]  12.6% (used 41156 bytes from 327680 bytes)
Flash: [=====     ]  54.3% (used 995549 bytes from 1835008 bytes)
Building /data/esphome-web-5c8418/.pioenvs/esphome-web-5c8418/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/esphome-web-5c8418/.pioenvs/esphome-web-5c8418/firmware.bin"], ["/data/esphome-web-5c8418/.pioenvs/esphome-web-5c8418/firmware.elf"])
Wrote 0x104770 bytes to file /data/esphome-web-5c8418/.pioenvs/esphome-web-5c8418/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 14.38 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esphome-web-5c8418.local
INFO  -> $$$$$$$$$
INFO Uploading /data/esphome-web-5c8418/.pioenvs/esphome-web-5c8418/firmware.bin (1001328 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esphome-web-5c8418.local using esphome API
INFO Successfully connected to esphome-web-5c8418.local
[06:55:43][I][app:102]: ESPHome version 2023.4.4 compiled on May  9 2023, 06:55:09
[06:55:43][C][wifi:505]: WiFi:
[06:55:43][C][wifi:363]:   Local MAC: $$$$$$$
[06:55:43][C][wifi:364]:   SSID: 'flme'[redacted]
[06:55:43][C][wifi:365]:   IP Address: $$$$$$$$$$$
[06:55:43][C][wifi:367]:   BSSID: $$$$$$$$$$$$$[redacted]
[06:55:43][C][wifi:368]:   Hostname: 'esphome-web-5c8418'
[06:55:43][C][wifi:370]:   Signal strength: -62 dB ▂▄▆█
[06:55:43][C][wifi:374]:   Channel: 1
[06:55:43][C][wifi:375]:   Subnet: 255.255.255.0
[06:55:43][C][wifi:376]:   Gateway: $$$$$$$$$$
[06:55:43][C][wifi:377]:   DNS1: $$$$$$$$$$$$$$
[06:55:43][C][wifi:378]:   DNS2: 0.0.0.0
[06:55:43][D][uart_debug:114]: >>> 15:45:20:00:03:7D:04
[06:55:43][C][logger:294]: Logger:
[06:55:43][C][logger:295]:   Level: DEBUG
[06:55:43][C][logger:296]:   Log Baud Rate: 0
[06:55:43][C][logger:297]:   Hardware UART: UART0
[06:55:43][C][uart.arduino_esp32:108]: UART Bus 0:
[06:55:43][C][uart.arduino_esp32:109]:   TX Pin: GPIO1
[06:55:43][C][uart.arduino_esp32:110]:   RX Pin: GPIO3
[06:55:43][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[06:55:44][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[06:55:44][C][uart.arduino_esp32:115]:   Data Bits: 8
[06:55:44][C][uart.arduino_esp32:116]:   Parity: NONE
[06:55:44][C][uart.arduino_esp32:117]:   Stop bits: 1
[... 112 lines cut ...]
[06:56:02][D][uart_debug:114]: >>> 15:45:20:00:00:3D:05
[06:56:02][D][uart_debug:114]: >>> 15:45:20:00:00:3D:05
[06:56:02][D][pulse_counter:174]: 'Pool Water Flow min': Retrieved counter: 0.00 pulses/min
[06:56:02][D][sensor:110]: 'Pool Water Flow min': Sending state 0.00000 L/min with 1 decimals of accuracy
[06:56:02][D][pulse_counter:180]: 'Pool Water Flow min': Total : 0 pulses
[06:56:02][D][sensor:110]: 'Pool Water Total': Sending state 0.00000 L with 1 decimals of accuracy
[06:56:02][D][uart_debug:114]: >>> 15:45:20:00:00:3D:05
[06:56:03][D][uart_debug:114]: >>> 15:45:20:00:00:3D:05
[06:56:03][D][uart_debug:114]: >>> 15:45:20:00:00:3D:05
[06:56:03][D][century_vs_pump:140]: Pump command 45 no response received - removed from send queue
[06:56:03][D][uart_debug:114]: >>> 15:45:20:00:03:7D:04
[06:56:03][D][uart_debug:114]: >>> 15:45:20:00:03:7D:04
[06:56:04][D][uart_debug:114]: >>> 15:45:20:00:03:7D:04
[06:56:04][D][uart_debug:114]: >>> 15:45:20:00:03:7D:04
[06:56:04][D][uart_debug:114]: >>> 15:45:20:00:03:7D:04
[06:56:04][D][pulse_counter:174]: 'Pool Water Flow min': Retrieved counter: 0.00 pulses/min
[06:56:04][D][sensor:110]: 'Pool Water Flow min': Sending state 0.00000 L/min with 1 decimals of accuracy
[06:56:04][D][pulse_counter:180]: 'Pool Water Flow min': Total : 0 pulses
[06:56:04][D][sensor:110]: 'Pool Water Total': Sending state 0.00000 L with 1 decimals of accuracy
[06:56:04][D][century_vs_pump:140]: Pump command 45 no response received - removed from send queue
[06:56:06][D][pulse_counter:174]: 'Pool Water Flow min': Retrieved counter: 0.00 pulses/min
[06:56:06][D][sensor:110]: 'Pool Water Flow min': Sending state 0.00000 L/min with 1 decimals of accuracy
[06:56:06][D][pulse_counter:180]: 'Pool Water Flow min': Total : 0 pulses
[06:56:06][D][sensor:110]: 'Pool Water Total': Sending state 0.00000 L with 1 decimals of accuracy
'

gazoodle, 2023-05-09:
Hmm. You don't appear to be getting any response to the commands being sent to the pump which would certainly give the "Unknown" sensor values. It looks like your pump isn't switched into Modbus mode (DIP 1 - ON, DIP 2-5 OFF), but your original issue looked like you had tried that combination. Basically your logs ought to have lines like this:
```
12:19:41 | [D] | [uart_debug:114] | >>> 15:45:20:00:03:7D:04
12:19:41 | [D] | [modbus:077] | Modbus user-defined function 45 found
12:19:41 | [D] | [century_vs_pump:203] | Read value 2600 from page 0, addr 3
12:19:41 | [D] | [sensor:110] | 'Pool Pump Controller Demand': Sending state 2600.00000 RPM with 0 decimals of accuracy
12:19:41 | [D] | [uart_debug:114] | <<< 15:45:10:00:03:A0:28:18:89
```

gazoodle, 2023-05-09:
Hmm. You said that the demand buttons worked (which makes sense because they are sent on the TX line, but the sensor values aren't working as there is no record of any response). This would seem to indicate that the device is in modbus mode, but not getting any return data. Maybe the RX line is damaged? Can you check that both wires have continuity between the pump and the ESP device?

fgamache1, 2023-05-09:
SOLVE::: :). That's in the code. I use the MAX485 and I forgot to add flow_control_pin to my code into modbus section. 

```
'
modbus:  
  id: id_modbus_3  
  flow_control_pin: 4
'
```
Thanks you so much for your help. 

Awesome code.

closed 2023-05-09 by fgamache1