gazoodle/CenturyVSPump #5: Power consomption

fgamache1, 2023-05-10:
Hello,

Do you know if it's possible to add the Power Consomption of the motor. In the documentation of regal Modbus I see in the sensor mapping page (Peak current in AMP) Have you try to get that ?

That would be really interesting information to have.

rx78nt1alex, 2023-05-10:
Im not the creator but I tried here but I think I was close.
[Pool pump speed control - easy way?](https://community.home-assistant.io/t/pool-pump-speed-control-easy-way/400695)

BShurilla, 2025-06-03:
Add this in your esphome yaml file
`# Live Wattage frm the motor
  - platform: centuryvspump
    name: "Pump Input Power"
    type: custom
    page: 0
    address: 0x05
    unit_of_measurement: "W"
    accuracy_decimals: 0`