gazoodle/CenturyVSPump #6: Code does not build with ESPHome v2023.5.1

qwest07, 2023-05-19:
Getting build error:

```
In file included from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp:1:
src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h: In constructor 'esphome::century_vs_pump::CenturyVSPumpSensor::CenturyVSPumpSensor(esphome::century_vs_pump::CenturyVSPump*, uint8_t, uint8_t, uint16_t, const string&, const string&)':
src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h:23:60: error: no matching function for call to 'esphome::century_vs_pump::CenturyVSPumpSensor::set_unit_of_measurement(const string&)'
                 set_unit_of_measurement(unit_of_measurement);
                                                            ^
In file included from src/esphome/components/sensor/sensor.h:5,
                 from src/esphome/components/centuryvspump/CenturyVSPump.h:7,
                 from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.h:3,
                 from src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp:1:
src/esphome/core/entity_base.h:82:8: note: candidate: 'void esphome::EntityBase_UnitOfMeasurement::set_unit_of_measurement(const char*)'
   void set_unit_of_measurement(const char *unit_of_measurement);
        ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/core/entity_base.h:82:8: note:   no known conversion for argument 1 from 'const string' {aka 'const std::__cxx11::basic_string<char>'} to 'const char*'
Compiling /data/spa/.pioenvs/spa/src/esphome/components/text_sensor/text_sensor.cpp.o
*** [/data/spa/.pioenvs/spa/src/esphome/components/centuryvspump/sensor/CenturyVSPumpSensor.cpp.o] Error 1
```

BravaCentauri83, 2023-05-20:
Same, throws error compiling here too.
[image]

kmycek, 2023-05-22:
Subscribed. Same issue as above.

PR #7 (2023-05-23): Remove unused constructor
This constructor is not used and you should not add name and UoM to the contructor anyway,

fixes #6
closed 2023-05-23 by gazoodle