gazoodle/CenturyVSPump #8: Pentair Wisperflo VST Control

jaysalazar, 2023-06-29:
I attempting to use this code to control my Pentair Wisperflo VST. It purchase the RS485 cable and attempted to communicate with it using an ESP32 connected to a MAX485 adapter however I get following logs. I have put the pump in External Control Only and swapped the Green and Yellow wires. 

[15:34:05][D][century_vs_pump:140]: Pump command 45 no response received - removed from send queue
[15:34:06][D][century_vs_pump:140]: Pump command 45 no response received - removed from send queue
[15:34:12][D][century_vs_pump:140]: Pump command 43 no response received - removed from send queue
[15:34:13][D][century_vs_pump:140]: Pump command 45 no response received - removed from send queue

I need to confirm that data is being sent from the pump to the 485 adapter, which I try and confirm later tonight. 

Is this pump compatible with this code? Here is the info on the pump.  
[image]
[image]

GraceO77, 2024-04-04:
@jaysalazar did you ever manage to get this working? I have done the same thing and get the same error codes

jaysalazar, 2024-04-05:
@GraceO77 I did not. I ended up using nodejs-pool controller. It's been working pretty well.

avbdr, 2024-04-12:
@gazoodle seems doesnt work on Hayward VSP too. I were trying to replace my home grown implementation, but no luck. 
The protocol im using now seems different and 1st dip switch is off, so no 12v.
Protocol seems like that:
        // [START OF PACKET BYTES]: 2 byte. fix bytes which indicates Start of Packet [0x10 0x02]
        // [ACTION]: 1 byte. action 12/0x0C -- set RPM,  action 1/0x01 -- set remote control
        // [SOURCE ADD]: 1 byte. Byte Source Address [0x01] Source address can be 12(broadcast) in some operations
        // [DESTINATION ADD]: 1 byte. Byte Destination Address [0x0C]
        // [DATA] : 1- byte Data - RPM Data in Parentage 0% - 100% [0x64]
        // [CHECKSUM] : 2 Byte Checksum [0x00 0x83]
        // [END OF PACKET BYTES]: 2-bytes fix bytes which indicates END of Packet [0x10 0x03]
        // Example (address 0x00: all dip switches off (Pool Filter), address 0x02: 2nd dip switch on (Aux 1)) 
        // on 45%  [0x10, 0x02, 0x0C, 0x01, 0x00, 0x2D, 0x00, 0x4C, 0x10, 0x03]
        // on 100% [0x10, 0x02, 0x0C, 0x01, 0x0C, 0x64, 0x00, 0x83, 0x10, 0x03]
        // off     [0x10, 0x02, 0x0C, 0x01, 0x00, 0x00, 0x00, 0x1F, 0x10, 0x03]

closed 2026-08-19 by jaysalazar