gazoodle/CenturyVSPump #9: Replacement for EOL M5Stack ATOM ESP32 device device

Scope666, 2024-05-12:
Hello,

I'd love to implement your project to control my V-Green 2.7 HP VS pump.  It looks like the device mentioned here is EOL.  For the ESP side of it, it looks like the Atom by itself is still available:

https://shop.m5stack.com/products/atom-lite-esp32-development-kit

and they also have this:

https://www.mouser.com/ProductDetail/M5Stack/C124?qs=sGAEpiMZZMuqBwn8WqcFUipNgoezRlc4Khjk3aPR%252BA7kAFbHUB5BZA%3D%3D

For the RS485 side, they have this, which looks like it has the same chip as the one you used:

https://www.mouser.com/ProductDetail/M5Stack/A131?qs=Imq1NPwxi75XShTuCGBiEA%3D%3D

Just wanted to see what you thought might work.   Thanks in advance!!!!

gazoodle, 2024-05-13:
Ah yes, the device I linked to is EOL, but it’s just made up of https://shop.m5stack.com/products/atomic-rs485-base and https://shop.m5stack.com/products/atom-lite-esp32-development-kitI’ll update the readme. I think your challenge will be less of the electronics which are standardised and more on what the Modbus registers and comms style used by your pump. Access to a tech manual is going to be essential IMHO.

https://shop.m5stack.com/products/atom-lite-esp32-development-kit
https://www.mouser.com/ProductDetail/M5Stack/C124?qs=sGAEpiMZZMuqBwn8WqcFUipNgoezRlc4Khjk3aPR%252BA7kAFbHUB5BZA%3D%3D
https://www.mouser.com/ProductDetail/M5Stack/A131?qs=Imq1NPwxi75XShTuCGBiEA%3D%3D

Scope666, 2024-05-13:
Thanks so much for your reply.  I ordered the ATOM Lite and RS485 module separately, can't wait to test them out.

Luckily the 2.7 HP version uses the exact same commands, @BravaCentauri83 has the 2.7 like me and is using your code successfully.

rx78nt1alex, 2024-08-22:
I’m using an ESP32 with a buck boost converter to drop the voltage from 12 vdc to the 5 volts it needs in a weatherproof enclosure. I have a 6-7 inch wifi antenna to boost the signal out of it across the 70 feet to the house. It’s solid.

Scope666, 2024-08-22:
Just realized I never closed this.  I've been running this code successfully for months, it's rock solid!!!

closed 2024-08-22 by Scope666