gazoodle/CenturyVSPump #11: Work with ESP32, ESPHome, and Intelliflo VS?

jvallery, 2024-10-21:
Hi! 

I've been trying to get this to work with an ESP32, ESPHome, and on my Pentair Intelliflo VS pumps. I've tried both the MAX485 and the XY-017 transceiver with no success. I can see the TX, but the pump never replies. 

Has anyone had success getting this to work with these pumps?

-JV

poldim, 2024-12-30:
+1

Would love to see support for this pump

XMaarten, 2025-04-06:
I forked [nicostrown's code](https://github.com/nicostrown/ESPHome-Pentair-Intelliflo) and was able to get it to work by using [some lambda](https://github.com/XMaarten/ESPHome-Pentair-Intelliflo/blob/main/pentair-intelliflo.yamll). I am able to set the program and see the speed and power usage now.

poldim, 2025-04-08:
> I forked [nicostrown's code](https://github.com/nicostrown/ESPHome-Pentair-Intelliflo) and was able to get it to work by using [some [...]

Awesome.  Can you show your wiring schematic and hardware?

XMaarten, 2025-04-08:
I used the same device as this project, the [M5Stack ATOM ESP32](https://shop.m5stack.com/products/atom-rs485-kit). Connected only the A and B wires for RS485, and used a USB power adapter for power.